VoronDesign/VoronUsers #118: E3v6 compatibility question

cskozlowski, 2020-07-05:
I currently have a E3v6 hotend.  When I fit the printhead_rear_e3dv6.stl (standard voron 2.4 part) to the duct.stl  (part of this build), the cutout for the e3v6 hotend is not correct.  Looks like the duct.stl part might be ready for a mosquito hotend and not the E3v6.

FHeilmann, 2020-07-05:
@WarriorOfWire thoughts?

cskozlowski, 2020-07-05:
In the meantime,  I downloaded the duct.stl file into Fusion360 and modified it to support the e3v6 hotend.  Printing it and will check fitment later today.

WarriorOfWire, 2020-07-10:
sure, cad is in this pr:  https://github.com/VoronDesign/VoronUsers/pull/120

FHeilmann, 2020-07-10:
Can you also fix this issue in that pr please?

WarriorOfWire, 2020-07-10:
No, I don't have a v6 to test with.  Would need to modify the printhead_front_e3dv6.stl part.  I also don't have a Dragon to test its own separate special part.  This is a feature request, not an issue - and cad was requested as one possible resolution, which is at least in that pr.

Might be best to consider a revision to the afterburner to decouple hotend mounting from layer fan ductwork.

FHeilmann, 2020-07-10:
Be that as it may, if there is a non-functional part in your mod I'd rather remove it for now. Otherwise more people might print it and end up with a non-functioning printer due to part incompatibilities.

cskozlowski, 2020-07-10:
I designed a e3v6 part that works with your design.  I also redesigned the duct and fan cover (lower two parts) so there is no overhang issue.  Let me know if you want me to send the *.stl files your way.  I have it installed and it does function.  In your readme, might want to make sure folks need to change out their print fan (40mm x 40mm x 10mm).  It works well for ABS but the fan is really underpowered for PLA from my initial testing.

WarriorOfWire, 2020-07-10:
> could you please at least mention it in the readme.

It's the first thing in the BoM in the PR.  I'll put it in the intro.

> might want to make sure folks know that they need to change out the print fan  

It's in the BoM.  I'll write a brief introduction at the top to clarify.

>  It works well for ABS but the fan is really underpowered for PLA from my initial testing 

Good to know.  I print pla very rarely and do not need the large 2.4 fan 99%+ of the time.  I'll mention it in the intro.

Feel free to sling me your stls on Discord.  I'll take a look; some of the design was pretty intentional & I'm curious what you changed.

FHeilmann, 2020-08-02:
@cskozlowski did you get in touch on discord? If so I think we can close this issue, as the appropriate notes have been added to the files in a PR thats currently under review.

cskozlowski, 2020-08-02:
Agreed, if the PR is currently under review.  Close it.

closed 2020-08-02 by FHeilmann