VoronDesign/VoronUsers #128: Afterburner files available on request

mateuszbrzezinski, 2020-07-28:
"Afterburner files available on request" How can I get Afterburner files for 2.4?

spott, 2020-07-30:
what files are you looking for?

The f3d files for 2.4 contain all of afterburner.

mateuszbrzezinski, 2020-07-30:
Sorry, I though it will be linked to what I meant.
I meant this: https://github.com/VoronDesign/VoronUsers/tree/8f43acee8ba7549add9e005fe9325dc8d10554b6/legacy_printers/printer_mods/Slidr/PanzerChain

there is written: (Afterburner files available on request)

spott, 2020-07-31:
Ah.

I would hop on discord and ping that user (Slidr), he would be the one to ask.

FHeilmann, 2020-08-02:
@Slidr666 your presence is requested :)

Slidr666, 2020-08-09:
Actully readme seems to be outdated there.  There are already Afterburner Beta files available.

FHeilmann, 2020-08-25:
@mateuszbrzezinski I think this issue is "resolved" now right? I hope you've gotten the parts you've needed from Slidr on discord. If not, feel free to reopen this issue.

closed 2020-08-25 by FHeilmann