VoronDesign/VoronUsers #152: Hour counter interference on 250mm version

hernsl, 2020-09-11:
The corner of the hour counter does not sit flush with the skirt because the corner is obstructed by the raised curved surface of the skirt.

FHeilmann, 2020-09-15:
@MarcPot can you look into this please?

hernsl, 2020-09-15:
I wanted to add additional detail. It may be appropriate to close this ticket, may not be. At the least, additional specifications for the hour counter are required.

I purchased a counter only to find it was VAC as opposed to DC. I ordered a DC one on Amazon and it fit without problem. The AC one did not work. Both look nearly identical, except the AC version has slightly larger mounting flanges with less of a perimeter fillet, thus it's too large for this mod.

It could be that physical difference will always exist between VAC and DC versions. Unknown at this time. At any rate, I've got it working with the DC version and I love this mod. Thank you.

MarcPot, 2020-09-15:
I was planning to edit the mod anyways with a crude sketch and klipper configs, so i'll update the CAD anyways at the same time. Good that you had another option, saves some wasted plastic!

MarcPot, 2020-09-22:
@FHeilmann I fixed it in https://github.com/VoronDesign/VoronUsers/pull/158

In the new version the hour counter overlaps the plug panel. But this was needed to get enough clearance, I also added a warning in the readme.md for this.

FHeilmann, 2020-09-29:
I just merged #158. FYI @hernsl.

closed 2020-09-29 by FHeilmann