VoronDesign/VoronUsers #409: Ender 3 V2 Compatibillity?

samirthegreat, 2021-08-19:
Is this mod compatible with ender 3 v2

Ramalama2, 2021-09-25:
The voron 2.4 mod 🤣

strayr, 2021-11-15:
I mean if you print an voron carriage for an ender and use one of the supported hotends, then yes, if you want to use a creality heatsink for some odd reason, then no you need to find someone to do that...

Do I have a voron carriage for ender3v2? yes, am I modding this to fit my creality heatsink? yes. Why? I have a bunch of other printers and like the modularity of the toolhead. And an old slice heatbreak that fits my creality heatsinkthat I want to use. Ender carriage is up on thingiverse, creality mod is currently printing. Hope this helps.

https://www.thingiverse.com/strayr/designs

FHeilmann, 2021-12-06:
@samirthegreat did you find what you're looking for?

raymondh2, 2022-02-07:
closing due to inactivity

closed 2022-02-07 by raymondh2