VoronDesign/VoronUsers #538: Help：The temperature of the extruder header  can not rised  190+ °C

knight350147, 2021-11-28:
Hi guys:
    When I test my machine(Voron 2.4 Extruder PID test),The fludd always reports errors like the picture.And the temperature freezd 190°C  ：

[image]

The printer.cfg
[image]

Board：fysetc spider
klipper+fuidd

Who can help me?

knight350147, 2021-11-28:
remark：sensor type like this：
[image]

FHeilmann, 2021-12-06:
This is not the place to ask for priner supprot, please join our discord or use our reddit instead.

closed 2021-12-06 by FHeilmann