VoronDesign/VoronUsers #540: Sizing of the heatserts holes in the spacelab2021/Orbiter-Clockwork

jtrmal, 2021-12-01:
Hi, the two "upper" heatsert holes seem to be a tad bit large -- the voron-standard heatserts require no force at all to be inserted. I see the specced ones would not have this problem, but are they really crucial? The other (4.7mm) holes are ok for the "voron-standard" heatserts
[image]

/cc @spacelab2021 @KiloQubit

jtrmal, 2021-12-01:
If there is any interest, I can provide "fixed" version. Also can provide an igus chain anchor model.

spacelab2021, 2021-12-10:
A lot of the issues have been addressed in my own repo and I haven't had a chance to update it in the voronusers repo. I should update the users repo shortly with the issues previously fixed.

jtrmal, 2022-03-22:
@spacelab2021 if you are too busy, can I help somehow? I will gladly do cleanup or any other work that needs to be done to finish this.

spacelab2021, 2022-04-05:
I've done all the modifications to the v1.5 orbiter Afterburner mount, but i still need to update it here. Thanks for reminding me. I've also been working on a orbiter v2 mount so I'll update that as well.

FHeilmann, 2023-07-07:
Closing due to lack of activity. Please reopen if any new developments arise.

closed 2023-07-07 by FHeilmann