VoronDesign/VoronUsers #552: Fluidd - Can't upload gcode files to print - "root is not available"

tlgrimmy, 2021-12-24:
I'm really new to all of this, so I apologize ahead of time if I'm posting things in the wrong spot or not including key info. I just put my Voron 2.4 together and managed to hack my way through the Klipper and Fluidd installs. My latest issue is that I can't seem to upload and print anything. I assume you upload by going to the Jobs tab and clicking a "+"  to upload a new file, just like I did with my printer.cfg file. However, there's no "+" sign on the jobs tab, just an red error triangle in its place that carries the message "No root available. Check your logs." I looked in the Moonraker log and it state's the following:

2021-12-24 00:03:43,148 [web.py:log_exception()] - 400 POST /api/files/local (192.168.XXX.XX): Root gcodes not available
2021-12-24 00:03:43,151 [app.py:log_request()] - 400 POST /api/files/local (192.168XXX.XX) [_TRUSTED_USER_] 312.51ms

I put Xs in for the IP address, but I think that's just a local IP and of no concern. I just don't know this stuff well enough to be sure whether I should have left that there or not. I don't know what any of this means, so I thought I would reach out for help. I've read a few posts about issues with permissions and incorrect directory pathways, but i just don't know for sure if it applies to my situation and don't have the slightest clue about how to change of it, ha! 

Any help would be greatly appreciated because, as of now, I just have a giant Voron 2.4 paperweight. Thanks!

[moonraker (1).log](https://github.com/VoronDesign/VoronUsers/files/7771881/moonraker.1.log)

tlgrimmy, 2021-12-24:
Figured out that this was a much simpler fix than I thought. I just had to add [virtual_sdcard] to the printer.cfg file. I had been ignoring dealing with those three errors that came up initially because a number of videos I had watched showed the same errors and they all mentioned they'd just deal with them later. Little did I know that you cannot upload print jobs without the [virtual_sdcard] code in place. Hope this helps someone else out there and Merry Christmas!

raymondh2, 2021-12-30:
This isn't the right place for fluidd support.

closed 2021-12-30 by raymondh2