VoronDesign/VoronUsers #585: Hidden Cable Z Cover Not 2.4 compatible.

Kevinthem0nk, 2022-01-20:
After tearing down and not really thinking about it I found that the cover can't fit under the 2.4's Z linear rails as I think is intended.  The 2.4 part and the dimension on the rail height in the manual are 3mm. I plan to edit the file for 2.4 builds in that area or maybe move the Z rails up 3mm to give them a small amount of clearance. Not really keen on moving the rails though. Please include a note of incompatibility with BOM/Stock build in the README. MD ... that being said, thank you for your hard work.

stefa168, 2022-09-22:
Do you have the edited file, if you got any further with the changes?

FHeilmann, 2023-07-07:
Closing due to lack of activity. Please reopen if any new developments arise.

closed 2023-07-07 by FHeilmann