VoronDesign/VoronUsers #593: LoudOwl aka Stabby airflow issue

gsdaemon, 2022-01-24:
@Baltojikale - first of all, thank you for the amazing mod! Everything was printing nice and fit.

Just one issue: Have you/anyone else made a water test for the fans? Both of my fans are spinning fine, but I have the impression that just one side is effective since the air is mostly blowing from the right hole near the nozzle.

Is this supposed to be like that or expected?

Thank you

gsdaemon, 2022-01-25:
I think the image below illustrates better the problem. Since the fans are supposed to spin and blow in one direction, but the shape of the casing inside the model is mostly mirrored, one of the fans' airflow will hit the X mark and have nearly no effect.

[image]

Baltojikale, 2022-01-25:
Yes the issue is known. left fan is sub optimal.  idea was that fan would still push out excess air but it seems to produce turbulence. model additional Impaler though it was designed to minimize size and use only one cutting tool. thus design has in perfections. after issue came out personally did not have time to access issue.
[image]

gsdaemon, 2022-01-25:
Thanks for confirming the issue. Maybe it's worth mentioning about this problem in the mod description to avoid people mutilating their fans before they realize one of them is not efficient.

xerootg, 2022-02-14:
fwiw - I took an exact-o knife, soldering iron, and file to mine today and ported the left fan cavity to see if I could improve the left fan. yes, you can, substantially so. I removed quite a lot of the material surrounding the m3 bolt shaft. I think with a little reworking of the left air channeling, this is quite an effective fan housing.

lisander-lopez, 2022-09-19:
Any updates on fixing this issue? My cad skills aren't that great for a piece like this!

FHeilmann, 2023-07-07:
Closing due to lack of activity. Please reopen if any new developments arise.

closed 2023-07-07 by FHeilmann