VoronDesign/VoronUsers #635: Decontaminator nozzle_clean script has a bug that affects right-side mounted brush/buckets.

LoganFraser, 2022-02-19:
The last X move in the nozzle_clean script is as follows: 
G1 X{bucket_left_width / 4} F{prep_spd_xy}

But for a right side mounted bucket, the line needs to be 
 G1 X{bucket_start + (bucket_left_width / 4)} F{prep_spd_xy}

This should also work fine with left side mounted buckets.

nknotts, 2022-02-22:
I can confirm this fix worked for me

rogery555, 2022-02-27:
You should explain what the bug is, I have a right sided mount bucket and I'm not sure what this is supposed to fix as I have no issues

LoganFraser, 2022-02-27:
> 

The intention of the final X move is to move to just inside the left edge of the left side of the bucket, and does so for a left-side mounted bucket, but crashes the nozzle into the klicky dock if using a right-side mounted bucket. The replacement line adds in the bucket offset from X0 so it correctly moves to inside the bucket for a right side mount.

nknotts, 2022-02-27:
I had the same behavior as @LoganFraser, the final move crashed the nozzle into my left-mounted klicky dock. With the replacement line, the final move parks the nozzle to the left edge of the bucket.

rlaferla, 2022-03-02:
Some context for the patch.  Please confirm this is correct:

`      ## Clear from area.
      M117 Cleaned!
      G1 Z{brush_top + clearance_z} F{prep_spd_z}

      {% if enable_purge %}

      ### BUG:  G1 X{bucket_left_width / 4} F{prep_spd_xy} 
      G1 X{bucket_start + (bucket_left_width / 4)} F{prep_spd_xy}

      {% endif %}

      ## Restore the gcode state to how it was before the macro.
      RESTORE_GCODE_STATE NAME=clean_nozzle`

LoganFraser, 2022-03-04:
Yes, that's correct.

claudermilk, 2022-03-07:
I just came to make the exact same issue request with the exact same solution. It works for me as well.

LongDono, 2022-11-08:
Looks like this issue should be closed. :)

closed 2023-04-03 by newphreak