VoronDesign/VoronUsers #643: Voron 2.4, Klipper, Octopus inductive prode help

morktista, 2022-02-25:
Can anyone please help me with a voron 2.4 inductive probe setup on a octopus?
I,m unsure on what to wire it as,
be it the probe port directly, with diode on pg15 or 11,
with a 4.7k across + and trigger on the probe port also what to do in the config file.
im getting devation from 0.01-0.06 its the only thing stopping me finishing the build,
also would the fact i plan to get a octopus pro board as the octopusi have it on loan help?

RALaRock, 2022-03-02:
I set it up using the diagram on page 237 of the R2 Voron manual. The port is PG15. I used:
## For Octopus recommended port with BAT85 diode
pin: ^!PG15 # ^ enables the pull up resistor, ! inverts the logic, PG15 is the port shown on the diagram on page 237.

morktista, 2022-03-03:
It's an octopus pro not an octopus, it's in the manual you don't need one

raymondh2, 2022-04-01:
This is covered in documentation we have

closed 2022-04-01 by raymondh2