VoronDesign/VoronUsers #656: The Sexbolt mod and automatic Z offset

NAPCAL, 2022-03-12:
The Sexbolt might need a spring under the plunger to provide enough force to push the filament off of the nozzle when doing an automatic Z offset. The microswitch in the Sexbolt has a very light activation force, so the leftover filament (even at filament printing temps) may activate the microswitch before the nozzle is in contact with the plunger.

lisander-lopez, 2022-09-19:
Yes!!! this has been affecting me while I print my SW parts. First layer is no longer "automated", have to fudge around with baby stepping.....

Maybe scrubber on the nozzle would work before Z_CALIBRATION ?

FHeilmann, 2023-07-07:
Closing due to lack of activity. Please reopen if any new developments arise.

closed 2023-07-07 by FHeilmann