VoronDesign/VoronUsers #685: Incorrect/incomplete stl files in repo

kaibsora, 2022-04-14:
In ultrawaffles e3 to switchwire conversion, there are a couple of issues with the stls. 

The left xz left bearing block in the repo does not match the cad file, and also does not align with the right bearing block due to being 5mm longer. the cad file is correct. 

A very minor issue, but there is no inclusion for "generic" drag chain mounting, only igus. I am planning on creating the bearing block to have support for it, but I think it should be mentioned. 

I'm creating this because I'm not sure if i could submit a pr to correct these changes or not. 

@UltraWaffles for awareness

UltraWaffles, 2022-04-15:
Thank you for the feedback, ill take a look

kaibsora, 2022-04-15:
I have created an stl for the generic mount of the left block, based on
your igus mount, if you would like it.

UltraWaffles, 2022-04-15:
that would be cool if you could share that real quick

kaibsora, 2022-04-16:
[Z_bearing block_left_generic.zip](https://github.com/VoronDesign/VoronUsers/files/8498930/Z_bearing.block_left_generic.zip)
Here it is.

UltraWaffles, 2022-04-16:
I updated the stls and added ur bearring block to the list. Did u make a z chain lower mount for generic chain by any chance?

kaibsora, 2022-04-16:
I did not, i was going to use the one from the sw conversion, but i havent looked into seeing if there is a difference. did you have to modify it in your build?

UltraWaffles, 2022-04-16:
I only have a 2020 extrusion igus version. The original iirc, sticks out a bit

kaibsora, 2022-04-16:
Ok. I'll update your version with the generic mounting holes, and post it
here.

kaibsora, 2022-04-16:
[[a]_z_chain_lower_mount_generic.zip](https://github.com/VoronDesign/VoronUsers/files/8499162/a._z_chain_lower_mount_generic.zip)

kaibsora, 2022-04-18:
I also just noticed that the x and z tensioners in your cad model are upside down. Not a huge issue, but something worth looking into

kaibsora, 2022-04-19:
Also, there are apperantly 3 cuts in at least your left deck dxf and 2 cuts in your right dxf. I don't know how to fix that or if that's normal or not.

FHeilmann, 2023-07-07:
Closing due to lack of activity. Please reopen if any new developments arise.

closed 2023-07-07 by FHeilmann