VoronDesign/VoronUsers #725: Rapido mount seems wrong dimensions

ocgeek, 2022-05-11:
Hi,
the mount that is published for Rapido appears to be incorrect. The nozzle is too close to the fan shroud.
I sent a message in the discord channel but it has been apparently disregarded.
If so there is a good chance to crash the fan shroud into the bed surface while homing Z...
[image]

ocgeek, 2022-05-11:
https://github.com/Badnoob-327/VoronUsers/tree/Rapido_abbn_30
@Badnoob-327

AaronHerrick, 2022-05-26:
I'll take a look

FHeilmann, 2023-07-07:
AB-BN has since been incorporated into the Voron Ecosystem as part of the SB DNA. Closing due to lack of activity. Please reopen if any new developments arise.

closed 2023-07-07 by FHeilmann