VoronDesign/VoronUsers #741: Mod incorrectly marked for V0

rapalax, 2022-06-05:
@samwiseg0 

According to the BOM for v0.1, The PSU should be a LRS-150-24.. 

The following mod is marked for compatibility for the V0.1 when there is no PSU cover for the LRS-150-24. 
https://github.com/VoronDesign/VoronUsers/blob/master/printer_mods/samwiseg0/lrs_screw_terminal_cover

samwiseg0, 2022-06-06:
Thanks for bringing this up. Can you do me a huge favor and print and test this? If this fits properly on a 150 I will PR it to the main repo. 

[lrs_150_screw_terminal_cover.stl.zip](https://github.com/VoronDesign/VoronUsers/files/8846458/lrs_150_screw_terminal_cover.stl.zip)

Thanks!

rapalax, 2022-06-11:
Tested - works like a charm! 

Thanks for doing a new version so quick.

FHeilmann, 2023-07-07:
Closing due to lack of activity. Please reopen if any new developments arise.

samwiseg0, 2023-07-07:
I will work on the PR this week. Sorry for the delay

FHeilmann, 2023-10-21:
@samwiseg0 Any Update on this?

PR #995 (2023-10-24): Move some mods to orphaned, clarify readmes, fix/add STLs
- Update `samwiseg0/lrs_screw_terminal_cover` with new files, (fixes #741, #829)
- Add note marking `togi/exhaust_housing_side_fitting` as explicitly incompatible with V2.4r2 (fixes #823)
- Move `42bios/LED_Mount` to orphaned mods (fixes #853)
- Move `Chaoticlab/Tool-free-tensioner` to orphaned mods (fixes #858)
- Export missing STL files for `Maverick/V0.2_Stealth_Skirts` (fixes #947,  #975)
- Add note to `alanho/PITFT50_v2_Mount` to explicitly mention that this mod is meant for the PiTFT50, add CAD (fixes #956, #993)
- Move `Maverick/ [... cut]
closed 2023-10-24 by FHeilmann