VoronDesign/VoronUsers #749: Voron0_MGN9C_X_Axis Carriage Bumps A Motor

adawalli, 2022-06-24:
Really love the MGN9C mod for the V0 by @hartk1213 

[printer_mods](https://github.com/VoronDesign/VoronUsers/tree/master/printer_mods)/[hartk1213](https://github.com/VoronDesign/VoronUsers/tree/master/printer_mods/hartk1213)/Voron0_MGN9C_X_Axis/

However, even with the carriage provided in this mod, the carriage is touching motor a which prevents the Y Endstop from closing. 

[image]

Any thoughts on improvements @hartk1213 ? Thanks again for this beautiful mod.

newphreak, 2022-07-24:
https://docs.vorondesign.com/community/howto/clee/sensorless_xy_homing.html might be worth checking out

adawalli, 2022-07-24:
> https://docs.vorondesign.com/community/howto/clee/sensorless_xy_homing.html might be worth checking out

That's actually what I've been using, however I think this mod should at least document this behavior or modify the a motor

CastaBrice, 2022-08-30:
https://github.com/nhchiu/VoronMods/tree/main/V0/X_axis_MGN9C
You can print drive frames from this repo

FHeilmann, 2023-07-07:
Closing due to lack of activity. Please reopen if any new developments arise.

closed 2023-07-07 by FHeilmann