VoronDesign/VoronUsers #754: Front fan clearance issue with abbn30 and lgx

Urminme, 2022-07-04:
I have an LGX installed on my trident and have a voron revo, but previously had a dragon which was fine but with the new voron Revo it seems like there is a few millimeter gap on the fan shroud where it meets the duct, I’m not sure if this is a known issue but I did not see anything about it yet 
[image]

>

newphreak, 2022-07-24:
did you remove the bondtech sticker? It's known to push things forward.

Urminme, 2022-07-24:
Yes I did, I ended up sanding down the spacer which seemed to help, I
triple checked I had all the correct updated parts as well just to be sure

FHeilmann, 2023-07-07:
Closing due to lack of activity. Please reopen if any new developments arise.

closed 2023-07-07 by FHeilmann