VoronDesign/VoronUsers #782: wrong STL file for 2.4R2 Pinned version in Voron2.4_Y_Endstop_Relocation

Blinker73, 2022-08-06:
Hello, 

I wanted to use 2 mods from hartk1213 that change the A motor bracketon a V2.4R2

Pins Mod
https://github.com/VoronDesign/VoronUsers/tree/master/printer_mods/hartk1213/Voron2.4_Trident_Pins_Mod
Y endstop Mod
https://github.com/VoronDesign/VoronUsers/tree/master/printer_mods/hartk1213/Voron2.4_Y_Endstop_Relocation

My PIF provider pointed out a mismatch between both sides of the A motor frame.

As per the links in the Y mod's page, I had provided him these files
The upper part from the Y endstop mod
(File with off centered hole) https://github.com/VoronDesign/VoronUsers/blob/master/printer_mods/hartk1213/Voron2.4_Y_Endstop_Relocation/STLs/Gantry/AB_Drive_Units/a_drive_frame_upper_with_jst_y_endstop_pinned.stl
The lower part from the Pins mod
https://github.com/VoronDesign/VoronUsers/blob/master/printer_mods/hartk1213/Voron2.4_Trident_Pins_Mod/STLs/Gantry/AB_Drive_Units/V2/a_drive_frame_lower_pinned.stl

It seems the upper part relocates the X bracket screw hole off center while the lower part has no relocation ending up as a half moon hole when the parts are joined.
After asking for help on Discord, a user named garbqgebag V2.1135 V0.610 pointed me to the mod's author's GitHub an to this file wich resolves the problem.
(corect file with centered hole) https://github.com/hartk1213/MISC/blob/main/Voron%20Mods/Voron%202/2.4/Voron2.4_Pins_Mod/STLs/Gantry/AB_Drive_Units/a_drive_frame_upper_with_jst_y_endstop_pinned24r2.stl

Can we include the correct file in the Y mod's git page.

Aslo, it seems the non pinned version also has the issue of the off center hole but don't know where to find a proper unpined file for the 2.4r2

At the top is what can be downloaded now.
at the bottom is what it should look like (red arrow)

[image]

Thanks

MapleLeafMakers, 2022-08-06:
@hartk1213 any chance you can open a PR with that change?

hartk1213, 2022-08-06:
> @hartk1213 any chance you can open a PR with that change?

Yeah not at home currently can when I get home in a few days

bassco, 2022-09-18:
for reference: https://github.com/VoronDesign/VoronUsers/pull/817 has the corrected files

newphreak, 2022-09-24:
> @MapleLeafMakers <https://github.com/MapleLeafMakers> PR with the files
> to resolve this issue has been made should we close this issue or wait
> until the PR is merged?

I think as long as it's linked like it gets when mentioned, it can be
closed automatically when merged.

closed 2023-07-07 by FHeilmann