VoronDesign/VoronUsers #824: Bed Fans - FansFast also based on Chamber temperature threshold

jeffhengesbach, 2022-09-16:
I've often observed my bed will reach target temp and the bed fans adjust to their fast setpoint as expected.  The issue is the chamber air is still cool enough the fans outrun the bed heater.  Some options for this are to pre-heat at a bed temp below threshold of the fan logic(no fans), manually adjusting the fan speed (too much effort), or setting a lower fast speed value  so the fans won't outrun the bed heater.  A reduced fast setpoint however compromises on achieving higher chamber temps.

I've added logic to the macros that will also consider a chamber thermistor temperature before putting the fans into the fast speed.
 In this way the bed fans will stay at the slow setpoint until both the bed has reached target (-1) and the chamber temp threshold.  It allows for a higher 'fast' fan speed that won't outrun the bed heater due to the warmer air and ultimately achieving higher chamber temps.

[bedfans.cfg.txt](https://github.com/VoronDesign/VoronUsers/files/9579652/bedfans.cfg.txt)

newphreak, 2022-10-09:
@AndrewEllis93

AndrewEllis93, 2022-10-13:
Just acknowledging this, I will take a look when I have some time

AndrewEllis93, 2022-12-29:
@jeffhengesbach Someone showed me that you can actually set your bed fans up as a fake "chamber heater". Seems to work really well, actually, thought you might be interested. 
See [here](https://github.com/AndrewEllis93/v2.247_backup_klipper_config/blob/master/bedfans_test.cfg) for my current test config. Adds M141 and M191 gcodes to set a target chamber temp, and the fans will regulate speeds to that setpoint

newphreak, 2023-04-03:
Closing as it's answered.

closed 2023-04-03 by newphreak