VoronDesign/VoronUsers #844: Obsolete link for component

flow23, 2022-10-19:
Hi,

the link to the hour counter from AliExpress is not working anymore.
Maybe this is the replacement: [AliExpress](https://www.aliexpress.com/item/32952929511.html?spm=a2g0o.productlist.main.7.444a7927NUILsv&algo_pvid=d4ae6750-bb90-421e-be7d-fbb5670b8438&algo_exp_id=d4ae6750-bb90-421e-be7d-fbb5670b8438-3&pdp_ext_f=%7B%22sku_id%22%3A%2266378730341%22%7D&pdp_npi=2%40dis%21EUR%215.91%215.91%21%21%213.15%21%21%402100b5dc16661827500314822d073a%2166378730341%21sea&curPageLogUid=Ij4PgnrXHIOR)

Link to mod:
https://github.com/VoronDesign/VoronUsers/blob/master/printer_mods/MarcPot/Skirt_Mods

Best regards,
Florian

newphreak, 2022-10-28:
@MarcPot

MarcPot, 2022-10-28:
Not planning to update/review my mod.
Srry

cbc02009, 2022-11-01:
@flow23 the mod is now community maintained, so you're welcome to submit a PR with a fixed link.

newphreak, 2022-11-02:
if you do, mention `#844` in the description or a comment so this issue gets linked and closes when it gets merged.

FHeilmann, 2023-10-21:
Closing, as mod is now community maintained and awaiting contribution from community

closed 2023-10-21 by FHeilmann