VoronDesign/VoronUsers #858: Tool Free Tensioner MOD - STL/CAD differes, STL dimensions are out of spec

hadesfactor617, 2022-11-09:
After printing the tool free tensioner, the pulleys would barely fit into the tensioner.  After taking measurements of the original voron Z-idler tensioner, I noticed the dimensions where 1mm off.  I was going to just edit the CAD until I noticed the CAD is completely different geometry then the STL posted.

Also, the README calls for 4x the hardware for 4 idlers, and to print 4 wheels but says to print only 2 tensioners and 2 brackets.  Seems as though there should be 4xbracket and tensioners but also an A/B side

top= stl, bttm=CAD

Blue tensioner=MOD, Black=Orig Voron2.4r2

[image]
[image]

Not sure if this was designed for the R1 in mind 

[image]
[image]

RainerBetrug, 2022-11-27:
[Z Achse Gurtstraffer.zip](https://github.com/VoronDesign/VoronUsers/files/10099165/Z.Achse.Gurtstraffer.zip)

newphreak, 2023-04-03:
@Chaoticlab: is this something you could answer?

PR #995 (2023-10-24): Move some mods to orphaned, clarify readmes, fix/add STLs
- Update `samwiseg0/lrs_screw_terminal_cover` with new files, (fixes #741, #829)
- Add note marking `togi/exhaust_housing_side_fitting` as explicitly incompatible with V2.4r2 (fixes #823)
- Move `42bios/LED_Mount` to orphaned mods (fixes #853)
- Move `Chaoticlab/Tool-free-tensioner` to orphaned mods (fixes #858)
- Export missing STL files for `Maverick/V0.2_Stealth_Skirts` (fixes #947,  #975)
- Add note to `alanho/PITFT50_v2_Mount` to explicitly mention that this mod is meant for the PiTFT50, add CAD (fixes #956, #993)
- Move `Maverick/ [... cut]
closed 2023-10-24 by FHeilmann