VoronDesign/VoronUsers #887: Issue with Angry cam freeing from mounting blocks

Eremius, 2023-01-06:
I have printed this part a couple of times, is there a recommended layer height to not have the camera holder stick to the mounting blocks?

chri-kai-in, 2023-02-05:
Hello Eremius,

a certain stick between mounting blocks and camera housing can happen, but should be easy to remove with medium manual force applied by moving the housing around the bolt of the mounting block. The bolts will not be 100% circular, which is OK, as the friction will allow to position the camera later at your desired angle, but still holds it in place safely.

I have not facing the issue of severe "glueing together" of those parts, when following the standard voron printing parameters, as mentioned in the voron build manuals:

Recommended Material: ABS
Layer Heigth: 0,2mm
Extrusion Width: 0,4 mm
Infill Ratio: 40%
Infill Type: Grid, Gyroid, Honeycomb, Triangle or Cubic
Wall Count: 4
Solid Top/Bottom Layers: 5

Printing should happen "face down", as provided in the models uploaded to GitHub.

If all of your other prints show perfect print quality, i would be interested in your print parameters and the type and brand of filament you are using. Maybe i can reproduce the situation on my printer to check on possible counter measures in my models.

Else, if you face severe sticking between the parts, which you can not loosen with manual force applied, there might be several root casues for this:
- check X/Y/Z dimension accuracy, measure the mounting blocks and compare to the dimensions in the 3D model. In case the prints are too big, enhance the value for "Shrinkage" in the filaments section of your slicer settings.
- your printer could apply too much material to the perimeters - check your pressure advance settings and/or the calibration of your extruder or the extrusion ratio of your filament
- check the cooling for material overhangs - it might be too low, which will influence the print quality on the overhang areas of this print e.g. at the bolts of the mounting blocks.
-...

I hope i can help a bit with my comments. Unfortunately there is a lot of root causes possible for such result.

BR

cscashby, 2023-02-20:
I've got very similar issues. Two different printers (an ER20 and a Voron 350), Good PLA+ (Eryone) filament which I've used with success to print a number of other things. Voron tuned to print ABS Voron parts to good dimensional accuracy.

To make it easier for people struggling with this, is it possible to provide the same model / STL as three separate parts? Left Right blocks and the main camera tube? If the blocks are screwed to the extrusion and press tight either side of the tube then it would still enable it to be mounted well without any issues.

cscashby, 2023-02-21:
I've now had some success by printing at 104% - they're very stiff to separate with two sets of pliers, but it can be done.

The camera works great and I've found that I can cut another - SHIELDED - USB cable and join to the original near the heatshrink to extend without signal drop issues, providing the shielding is folded back over the joined and heatshrunk individual cores.

newphreak, 2023-04-03:
Sadly somewhat common it seems. Hopefully you'll find something that works long term. I'm closing this one now then.