VoronDesign/VoronUsers #910: Hidden cable routing z belt cover mod not fitting when following r2 build instructions

scheini79, 2023-02-15:
When attaching the Z rails following the build manual for r2 it suggests a gap of approx. 3mm (Page 25). Hidden cable routing z belt cover mod is 5mm in height, thus not fitting under the rails. I did a remix to get rid of the problem.

I'm not sure if this is the right place for a remix, i'm fairly new to GitHub, so please bear with me.
[image]
[image]
[cable_routing_z_belt_cover_a_mod v1.zip](https://github.com/VoronDesign/VoronUsers/files/10742445/cable_routing_z_belt_cover_a_mod.v1.zip)
 Attached you will find the .f3d with my solution.

jd766, 2023-03-26:
The files haven't been updated with the correction above, this issue still exists

newphreak, 2023-04-03:
does #872 have this fix, or should it combine it?

FHeilmann, 2023-10-21:
since this is an issue with a Base Voron part, please submit an issue to the appropriate printer repo if the issue still persists.

closed 2023-10-21 by FHeilmann