VoronDesign/VoronUsers #936: Z belt cover

Sidkrsk, 2023-04-14:
It would be good to make a groove under the Z linear guides. Since according to the instructions for voron 2.4 r2, the gap from the bottom is 3 mm, and the plug is 5 mm high and does not get up normally

https://github.com/VoronDesign/VoronUsers/tree/master/printer_mods/Akio/cable_routing_z_belt_cover

smplman, 2023-05-01:
I needed this also and took a stab at modifying the files. I have them printed and installed on my printer and they are working well. 

https://github.com/smplman/VoronUsers/tree/master/printer_mods/smplman/cable_routing_z_belt_cover_rail_gap/stl

FHeilmann, 2023-10-21:
@Akio-42  Is this something you could see yourself fixing in the near future?

Akio-42, 2023-10-21:
@FHeilmann Sure. Did not know about this issue as it was not linked with my current pull request. Will make the change and include it soon.

FHeilmann, 2023-10-21:
> @FHeilmann Sure. Did not know about this issue as it was not linked with my current pull request. Will make the change and include it soon.

Thanks! Please tag me in the PR once it's ready

Akio-42, 2023-10-23:
@FHeilmann tagged you at https://github.com/VoronDesign/VoronUsers/pull/872

closed 2023-10-24 by FHeilmann