VoronDesign/VoronUsers #959: 350mm issue with V2.4 Skirt switch mod by tayto-chip

TheSin-, 2023-06-10:
Sorry @tayto-gp there is no user with the name tayto-chip on discord which the readme says to contact with issues.

The 350mm side a skirt won't work the nub is too long.

[image]
[image]
[image]

tayto-gp, 2023-06-10:
Hello @TheSin-, I made several fixes to all of these a while back but uploading them has been put on the back burner. Sorry about the issues but thank you for reminding me to update github. In the meantime, if you contact me on discord (tayto #8140) I can send you the updated stl.

TheSin-, 2023-06-10:
Perfect thank you, message sent.

TheSin-, 2023-07-18:
this was fixed in the GitHub and I tested it, it's perfect now!

closed 2023-07-18 by TheSin-