VoronDesign/VoronUsers #991: Is this mod useable with Voron-1 (Trident) 300 which has rotated DIN rails?

dseligson, 2023-09-25:
The Voron Trident 300 has DIN rails which are rotated 90 degrees relative to the Voron Trident 250.  (attaching from front-back rather than side-side)
Will the same C-Clamps work in that orientation?

dseligson, 2023-10-21:
> @dseligson can you please specify which mod you're talking about?

https://github.com/VoronDesign/VoronUsers/tree/master/printer_mods/LoganFraser/TridentInvertedElectronics

FHeilmann, 2023-10-24:
@LoganFraser can you comment on this?

jwyse, 2023-10-31:
I the same question.  Additional context from [LDO's wiring guide](https://docs.ldomotors.com/en/voron/voron-trident/wiring_guide_rev_a#installing-the-din-rails) for their Trident kit:

> The orientation of the DIN rails in our guide differs from that of the official Voron manual. These DIN rails run from front to back.

[image]

I'm sure we could build the LDO kit with the orientation from the Voron manual rather than LDO's wiring guide, but might run into some clearance issues with the electronics and wire harnesses provided.  Input from others who have tried it would be helpful, before printing a bunch of parts.

jwyse, 2023-10-31:
@dseligson  It looks like it should work. I put together just enough to test-fit the deck panel on the C-shaped supports with rails front-to-back.  I had to push down on the deck panel in front of the rear Z stepper to gently flex the deck panel to get it under the front lip of the stepper mount.

[image]
[image]

LoganFraser, 2023-11-01:
You might want to check how the power inlet interacts with the C shapes.  It might be better to have the open side face towards the outside of the trident to avoid a conflict.  That's the reason for the slightly strange shape, so they can wrap around other inlets and work either on the sides and avoid blocking the fans or on the back and avoid the power and cable inlet panels.