VoronDesign/VoronUsers #993: BTT-PITFT50-v2-Mount - Needs Updating for V2.1

HasWalji, 2023-10-09:
@alanho 

Hi,
The current STL for the BTT BTT-PITFT50-v2-Mount is not compatible with BTT PITFT50 V2.1. The cutout doesn't line up with the buttons keping one button constantly pushed down.

[image]

Needs to be adjusted to support moving the cutout to the right to clear the buttons. See image atached.

Thanks.

WaschBaer0395, 2023-10-20:
Can confirm!, plus there needs to be a extra cutout inside the frame at the top of the screen (middle to top right when the buttons are facing down) for the lcd touch and backlight ribbons, on my 2.1 unit they stick out by almost a mm, and i dont wanna force them flush as the bend is pretty tight
for now i dremeled the cutout open and put a slot in said place

FHeilmann, 2023-10-21:
#956

It unfortunately doesn't look like the maintainer is willing to maintain this mod. Would you like to provide support for this mod by updating it?
@WaschBaer0395 @HasWalji

HasWalji, 2023-10-21:
I would have made an attempt, albeit a sad one but would have tried but the cad files aren’t there and just the STL’s. If the original 3mf files can be posted I think folks would be willing to update it.

alanho, 2023-10-21:
hey all, i don't (and won't) have a BTT PITFT50 V2.1 on hand for me to have an updated version of the mod. but i'm happy to upload the fusion 360 source file so anyone can make a update for it. will create a PR for the source file

WaschBaer0395, 2023-10-21:
https://www.printables.com/model/514367-tft-5-holder-for-voron-24-btt-pitft50-v21 i did find someone who seemed to have remixed this one ? labeled for v2.1  it has button cutouts and the cutout for the ribbon up top
altho the button cutouts are 2 big squares, it looks as if it might work for all 3 buttons when removing the center post

HasWalji, 2023-10-21:
I would go with this one as it has the cutouts exactly to match the buttons on BTT models and based on my test prints works quite well; however please do your own testing.
https://www.printables.com/model/537886-btt-pi-tft50-v21-screen-mount-for-voron-skirts_rem/files

WaschBaer0395, 2023-10-21:
> I would go with this one as it has the cutouts exactly to match the buttons on BTT models and based on my test prints works quite well; however please do your own testing. https://www.printables.com/model/537886-btt-pi-tft50-v21-screen-mount-for-voron-skirts_rem/files

i wouldnt recommend the one you posted, as it dosent have a groove cut at the top for the display ribbons
on some models they can stick out a little and bending them more could easily break them 
the one i posted has a groove cut at the top section to accomodate the inconsistent ribbons

HasWalji, 2023-10-21:
Alan was kind enough to post the fusion files; anyone else watching this repo who has experience with fusion and can update this model would be greatly appreciated. I’ll definitely give it a go however I am just learning fusion with simple models right now (full transparency).

WaschBaer0395, 2023-10-21:
> Alan was kind enough to post the fusion files; anyone else watching this repo who has experience with fusion and can update this model would be greatly appreciated. I’ll definitely give it a go however I am just learning fusion with simple models right now (full transparency).

both posted versions seem to be directly remixed from his model, they have the exact same dimensions/shape etc ( the one i poisted is just a bit taller to give a couple mm lip and not have the screen flush with the face)

FHeilmann, 2023-10-22:
> both posted versions seem to be directly remixed from his model, they have the exact same dimensions/shape etc ( the one i poisted is just a bit taller to give a couple mm lip and not have the [...]

Thanks for all your Investigation @WaschBaer0395 @HasWalji 

@alanho has provided his source file and I'll incorporate it as a part of several fixes in the repo soon. Unfortunately I can't just rip the printables parts from that website and include them here. So I am going to add a disclaimer and links to these models to the readme of alanho's Mod.

closed 2023-10-24 by FHeilmann