VoronDesign/VoronUsers #1004: RockNRoll Mod - Side Roller

leitenky, 2023-11-01:
Hi @RockNLol I was wondering if you would be able to add an option to to Side Mount your RockNRoll mod?
It looks like the only change needed would be to modify the two baseplate mods so the supports come out the side.

1x [a]_z_drive_baseplate_a_rnl_mod_r2
1x [a]_z_drive_baseplate_b_rnl_mod_r2

I could test and provide feedback if you'd like as well. Thank you for your time

RockNLol, 2023-11-01:
hi!
the mod should work without using the baseplate at all. During my testing period they were not part of the design. I have a pretty heavy 350mm 2.4, and they held up nicely. I added the baseplates for the release version more for safety and less flex in the part than for function.
Message me on discord, link is in the mods README.md. Maybe we can work something out, but my time is very limited at the moment.

FHeilmann, 2023-12-20:
@leitenky did you manage to resolve your query with RockNLol?

leitenky, 2023-12-20:
Yes I have.  I made some custom parts that's I've sent to him he will possibly update his mod with the additional files at some point but this issue can be closed.  Thanks

closed 2023-12-20 by FHeilmann