VoronDesign/VoronUsers #1032: [MOD ADOPTION] V2.4 Z Drive Motor Tensioner Mod

Dr-Info, 2024-01-22:
### Which mod do you want to adopt?

https://github.com/VoronDesign/VoronUsers/tree/main/orphaned_mods/edwardyeeks/V2.4_z_drive_motor_tensioner_mod

### Describe your plans for the mod

Would like to take over this mod. I modded the rear skirts and Skohls did the baseplates for V2.4R2.

### Have you contacted the original author?

- [ ] Yes
- [X] No
- [ ] Original author is not reachable

FHeilmann, 2024-02-03:
Hi, thanks for your interest in adpting this mod and sorry for the late reply. I hope I'll be able to get this process rolling soon. Are you planning on making any other updates to the mod while you're adopting it?

Dr-Info, 2024-02-06:
I don't think there are needed updates and I don't see any reported issues for the mod.

FHeilmann, 2024-02-18:
Sorry again for the delay, I have opened #1034 to facilitate the adoption. Feel free to add any additional information to the PR.

closed 2024-02-23 by newphreak