VoronDesign/VoronUsers #1040: [MOD ISSUE] Taille bouton mod xbst_ V2.4_Skirt_Buttons

Didier-T, 2024-03-14:
### Which mod does this issue apply to?

xbst_/V2.4_Skirt_Buttons

### Relevant VoronDesign printer

Sur ma V2.4

### Describe the bug

Les boutons mis en lien ne correspondent pas aux extrusions faite dans le modèle à imprimer.
Il faudrait soit des boutons de 12x12x7, soit passer les extrusions de 1.7 mm à 0.5 mm pour régler le problème.

The linked buttons do not match the extrusions made in the template to be printed.
It would either require buttons of 12x12x7, or pass extrusions from 1.7 mm to 0.5 mm to solve the problem.

FHeilmann, 2024-03-14:
@xbst 
Anything we can do here?

xbst, 2024-03-14:
Please correct me if I'm misunderstanding your question, a lot seems to be lost in translation, but I think you're saying the hole in the 3D printed buttons is too deep for 12x12x5mm tactile switches, and you think they should be shallower or use 12x12x7mm tactile switches?

The parts in the readme are based on the parts I used years ago when I built this mod. I no longer have this mod installed on my printer (did the Doom Cube mod) so I can't check their shaft length, but unless I made a typo, 12x12x5mm should be correct. I remember using short shafted tactile switches. 

You need as much of the of the tactile switch's shaft to fit in the button as possible, otherwise the button may not stay in place well, or worse, super glue might leak into the tactile switch when you're gluing it on the tactile switch. That being said, if you prefer a shallower hole on the buttons, the CAD model is in this repo. Should be a simple modification to make. Just make sure you don't make the buttons stick out of the skirts too much. The hexagon cutout for the button to sit in is pretty shallow since I wanted to not change the aesthetics of these skirts too much for this mod. So if you make the buttons completely stick out of the skirts, they may get caught on the haxagon walls on the skirts.

Didier-T, 2024-03-14:
> Veuillez me corriger si je comprends mal votre question, beaucoup de choses semblent être perdues dans la traduction, mais je pense que vous dites que le trou dans les boutons imprimés en 3D est trop profond pour les commutateurs tactiles de 12 x 12 x 5 mm, et vous pensez qu'ils devraient être moins profonds ou utiliser des interrupteurs tactiles 12x12x7mm ?
> 
> Les parties du fichier Lisez-moi sont basées sur les parties que j'ai utilisées il y a des années lorsque j'ai construit ce mod. Je n'ai plus ce mod installé sur mon imprimante (j'ai fait le mod Doom Cube), donc je ne peux pas vérifier la longueur de leur tige, mais à moins d'avoir fait une faute de frappe, 12x12x5mm devrait être correct. Je me souviens avoir utilisé des interrupteurs tactiles à tige courte.
> 
> Vous avez besoin que la plus grande partie possible de la tige de l'interrupteur tactile s'insère dans le bouton, sinon le bouton risque de ne pas rester bien en place, ou pire, de la super colle pourrait couler dans l'interrupteur tactile lorsque vous le collez sur l'interrupteur tactile. . Cela étant dit, si vous préférez un trou moins profond sur les boutons, le modèle CAO est dans ce dépôt. Cela devrait être une simple modification à apporter. Assurez-vous simplement de ne pas trop faire ressortir les boutons des jupes. La découpe hexagonale dans laquelle le bouton peut s'asseoir est assez peu profonde car je voulais ne pas trop changer l'esthétique de ces jupes pour ce mod. Ainsi, si vous faites dépasser complètement les boutons des jupes, ils risquent de se coincer dans les parois haxagones des jupes.

Vous avez très bien compris le problème.
Et pour être plus précis la tige du bouton poussoir ne touche pas le fond de l'extrusion.
L'extrusion fait 1,7mm alors que la tige du bouton poussoir ne fait que 1mm et le mécanisme du bouton poussoir à besoin de parcourir 0,4mm pour être actionné.

Je suppose en effet qu'il y a eu une confusion lors du référencement des boutons poussoirs et que les mieux adaptés sont ceux de 12x12x7

Merci pour le fichier étapes, il m'a permis de modifier de manière simple le fichier pour le rendre utilisable avec les boutons que j'avais commandé spécialement pour mettre en place votre mod 😉

J'aimerais juste que cette mésaventure n'arrive pas à d'autres utilisateurs qui tomberaient sous le charme de votre travail 🙂

newphreak, 2024-04-20:
So what remains here?